LaserWeb/LaserWeb4 #635: Generate gcode for lasers/machines which do not support laser intensity.

Rabter1, 2022-02-12:


Rabter1, 2022-02-13:
Hi @cprezzi thanks for the tip. 

This indeed works partially for me if i delete the S lines manually.
But this only works in `Laser Fill Path` but not in `Laser Raster` or `Laser Raster Merge`

When using `Laser Raster` the logic is always Laser ON and only S value is changed even when laser is supposed to be off.

Also i noticed, that the letter i can set for Laser intensity (Settings->GCode->Laser Intensity) is ignored in `Laser Raster` it is always the letter `S`

easytarget, 2022-02-13:
> Is this project still maintained?

Yes: and a new release is in the planning. but there is a way to go. And it focuses more on UI changes and updating to 2022 web standards, no substantial changes to gcode generation.

cprezzi, 2022-02-13:
@Rabter1 Raster operations use a different library than the vector operations and this library never supported the configuration options from the gcode tab. If you like to add this, startt @ https://github.com/LaserWeb/lw.raster-to-gcode.git

easytarget, 2022-02-13:
@Rabter1 ,
 as @cprezzi says, LW4 actually has (for historical reasons) several independent generators for different operation types, the raster generator has a different heritage than the path cutting generator, and the various mill generators also have their own code. The individual generators are run as web-workers and all live in the `src/lib` folder, along with their glue logic.
- I now know the code quite well, but I still struggle to remember which operations are associated with the various generators, and which settings they use. If you need any help navigating around this code please ask. 

Also watch out for the v4.1 changes as they arrive; there are changes to the default machine profile list (it is expanded and some additional info fields are added).

Rabter1, 2022-02-17:
Since I need to add probably one UI elemt in the settings for Mach3 machines, I will wait with my changes until i have 4.1.
This will most likely make things easier because of the merge conflicts.

easytarget, 2022-02-17:
The 4.1 branch is rebased and renamed; and I'm doing some testing on it [here](https://github.com/easytarget-org/LaserWeb4) and hope to have it in the main repo soon; I'm still feeling my way with releasing.

Rabter1, 2022-02-17:
Okay thank you!
I will follow [this guide](https://github.com/LaserWeb/LaserWeb4/wiki/How-to-Contribute) and fork branch 4.1. Merge conflicts should be minimal than.

Rabter1, 2022-03-27:
Today I had a little bit of time to code the changes. 
I created a [Pull Request](https://github.com/easytarget-org/LaserWeb4/pull/1)

easytarget, 2022-03-30:
> @easytarget could you give me a hint on how to create a .exe file?

I dont really know that myself yet; I have not yet got round to bundling my changes into the electron appimage. I am a bit stalled in the release process.

cprezzi, 2022-03-30:
The EXE is built from the electron_bundler branch of lw.comm-server (https://github.com/LaserWeb/lw.comm-server/tree/electron_bundler). Check out the scripts in package.json. The "nightlywindows64" tells you all the steps.