LaserWeb/LaserWeb4 #636: Path select in workspace via a region (select box)

easytarget, 2022-03-26:
Allow path selection in the workspace using a region defined via the mouse (area select? I dont know the correct UI term for this)
- This would require a modifier/specific tool since currently all the mouse buttons are used in the UI for pan/rotate/zoom
- It needs to allow for area deslection/selection-inversion too; this is typically done via a keyboard modifier.

(UI feature request from https://forum.makerforums.info/t/newbie-here-do-you-have-some-drag-select/85180)

harlock999, 2022-03-26:
Can someone label this as `feature request` , I don't seem to have the access rights to do so.

easytarget, 2022-03-27:
@cprezzi normally does this as necessary, we can follow up on MakerForums..