LaserWeb/LaserWeb4 #639: X, Y Shift after Z move in multi pass cut op.

opensourcefan, 2022-05-22:
A big THANK YOU to all for keeping this alive. Having a FOSS option is great.

***

More often than not I'll get a layer shift when I'm printing a larger multi pass **cut** file. 

I'm burning on an older 8bit converted 3d Printer running Repetier 0.92. gcode ops seem fine from what I can see. 

I have a new 32bit dedicated GRBL board coming. Just thought I would post this in case it's not just me.

cprezzi, 2022-05-23:
Lost stepps are in most cases caused by mechanical overload of the stepper motors or weak drivers, sometimes too hight accellerations. You might also have USB communication problems. In this case you should test shorter, high qualty USB cables.

Even 8 bit boards should not produce stepp losses, if they are correctly configured and the drivers can deliver enough current for the motors.

LaserWeb is not known to produce any "layer shifts". This would be visible in the generated gcode.

opensourcefan, 2022-05-23:
Okay cool, was just making sure. I'm not going to bother diagnosing the current board as it's old and not meant for the task. I'm sure the new board will fix the issue when it arrives. 

It's just strange that it happened in the exact same spot each time but that could be a USB buffer or something.

easytarget, 2022-05-25:
As @cprezzi says, the preview trace you see in the GUI is the path that will be sent to the controller.. so if the machine shows 'shifts' relative to that the fault must lie in the controller (or possibly something sticking / mis-aligned in the mechanism). Over-optimistic acceleration or travel speed values may be another cause.

Good luck with the new board, once you have tried a 32-bit controller you'll never want to go 'back' ;-)

closed 2022-05-25 by easytarget