LaserWeb/LaserWeb4 #640: Documents not showing after import on workspace after extended use.

opensourcefan, 2022-05-22:
After performing multiple gcode generations with multiple files eventually a new document stops showing in the workspace.

Restarting fixes the issue. Sometimes but not always "reset"ing the workspace helps but a restart is more reliable. Occurs with SVG and PNG, haven't tried anything else.

Sometimes if I hit generate the gcode moves will show even though no image was previously displayed. Toggling "show document"  has no effect.

harlock999, 2022-05-23:
There is a lot of rework to the frontend being done by easytarget. Can you try to toggle the 'display cache performance' in the LaserWeb settings? What browser and version are you using?

opensourcefan, 2022-05-23:
Using 4.0.999 and it appears that it's just opens it's own window vs a browser window. Edge is my default though. 

I'll try the toggle next time I get it to happen.

opensourcefan, 2022-05-23:
Oh and I should mention that I am exporting the gcode only, not monitoring or running from LW. I don't have my or any machine connected.

opensourcefan, 2022-05-25:
Appreciate the response, thank you for all the hard work. As my username suggestions I really appreciate FOSS.

Running on Windows 10, latest build.

I've got in the habit of just restarting before each project now, seems to keep things stable.