LaserWeb/LaserWeb4 #641: Anycubic Pro support?

blackPantherOS, 2022-06-12:
It's prossible?

lsusb: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Device Firmware version 1.2.9

https://laserweb.github.io/LaserWeb4/ log: 

 ```
Connecting to Server @ localhost:8000
 Warning: Gcode out of machine bounds, can lead to running work halt
 Connecting Machine @ USB,/dev/ttyUSB0,115200baud
```
(nothing else)

v4.0.996-130-x86_64.AppImage version log:

```
 Connecting to Server @ localhost:8000
 Server connected
 Machine disconnected
 Serial ports detected: [{"comName":"/dev/ttyS0"},{"comName":"/dev/ttyS1"},{"comName":"/dev/ttyS2"},{"comName":"/dev/ttyS3"},{"comName":"/dev/ttyUSB0","manufacturer":"Silicon_Labs","serialNumber":"Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001","pnpId":"usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0","vendorId":"0x10c4","productId":"0xea60"}]
 Connecting Machine @ USB,/dev/ttyUSB0,115200baud
 Machine connected
 �Np|�_JH5�P��JpON6��Z�z
 %�h�onm7�nn|&~�6�H�X�J�~�Z%�]]�M�7��-*&�]MbO6d6�p|HH6f�J�JJ�8J�(K�J�Q�*:&H^|**J�Z�A�J�A�J%�J�X�:&�^>h�$p|4�=�=�Jh%�X��^>Y�N66�^^aHM�6�p|4�=HJ�%�I�=HJp|@_a�&&��$MM6l*I$M^j�6l*I�6XH>�4�&('�h*I�&(n~l*I�&n~l*X�J�Z�I�j�J�(%�8%�(�J߈X�*�MahJ�J�I�jN�X�*��N6�h$p|�
 N�X�:66l6N&^f�J�[�(%M0�[�(J-H*
 jO6�f�J�Z�J�85�HJp|4�%�J&�HJ�X�
 No supported firmware detected. Closing port /dev/ttyUSB0
 Machine disconnected

```

cprezzi, 2022-06-12:
What type of firmware is running on a Anycubic Pro? LaserWeb only supports gcode based firmwares (see: https://github.com/LaserWeb/LaserWeb4#supported-firmwares).
The cryptic text after "Machine connected" doesn't look right. It should be text onliy.

blackPantherOS, 2022-06-13:
> What type of firmware
theoretically this is a Marlin based firmware

> The cryptic text after "Machine connected" doesn't look right. It should be text onliy.
yes, little bit chaos output, same as in Lightburn

blackPantherOS, 2022-06-13:
finally, text data was extracted with manual device settings in LightBurn, Ignored autodetect and manual selected Marlin firmware  But does not work correctly with LigthBurn, wrong the start position, can't start laser, etc..

```
start
echo:V1.2.9
 1.1.0-RC8
echo: Last Updated: 2016-12-06 12:00 | Author: (Jolly, xxxxxxxx.CO.)
Compiled: Jan  6 2021
echo: Free Memory: 2718  PlannerBufferBytes: 1168
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E384.00
echo:Maximum feedrates (mm/s):
echo:  M203 X300.00 Y300.00 Z10.00 E80.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1500 Y1500 Z800 E10000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P1500.00 R10000.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.50 E5.00
echo:Home offset (mm)
echo:  M206 X0.00 Y0.00 Z0.00
echo:Z2 Endstop adjustment (mm):
echo:  M666 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P20.00 I1.00 D50.00
echo:  M304 P100.00 I0.23 D20.40
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:SD card ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
ok
echo:SD card ok
ok
```

cprezzi, 2022-06-13:
The cryptic text could mean that you have selected the wrong baudrate for your controller. It might be other than 115200.

blackPantherOS, 2022-06-27:
> The cryptic text could mean that you have selected the wrong baudrate for your controller. It might be other than 115200.

I've tested with other baud rate, this is results:

```
Connecting Machine @ USB,/dev/ttyUSB0,57600baud
 Machine connected
 No supported firmware detected. Closing port /dev/ttyUSB0
 Machine disconnected
```
:(

blackPantherOS, 2022-06-29:
> I just realized you tried it first with the web testversion

Ohh... because it was so handy :)

> If you have installed LW (lw.comm-server)

the latest git version won't work for me, cannot connect with web4. 

> you can edit config.js and make sure line 14 is:

[image]

1st simpli run
```
./server-linux
> 
(To exit, press ^C again or type .exit)
> 

```

2nd variant

```
 node ./server.js 
 
***************************************************************
        ---- LaserWeb Comm Server 4.1.000 ----        
***************************************************************
  Use  http://192.168.1.100:8000 to connect to this server.
 
* Updates: 
  Remember to check the commit log on
  https://github.com/LaserWeb/lw.comm-server/commits/master
  regularly, to know about updates and fixes, and then when ready
  update accordingly by running git pull
 
* Support: 
  If you need help / support, come over to 
  https://forum.makerforums.info/c/laserweb-cncweb/78
***************************************************************
 
Server binding to all local IP addresses on port: 8000

```

 netstat -ntap  | grep 8000

```
tcp        0      0 0.0.0.0:8000            0.0.0.0:*               LISTEN      1514068/node 
```

The Web4 log (seems not connected to local sever):

```
 Connecting to Server @ http://192.168.1.100:8000
 Connecting Machine @ USB,/dev/ttyUSB0,57600baud

```

closed 2024-05-31 by cprezzi