LaserWeb/LaserWeb4 #642: No supported Firmware Detected

ricardofrias, 2022-06-18:
I received my Sculpfun S9 yesterday, using a Mac I did engraving tests, everything came out perfect. 

Today I went to do the cutting tests and the LW simply does not connect to the machine. I've already tried reinstalling the CH34 driver, the LW and nothing works. 

Does anyone know what could be happening?

cprezzi, 2022-06-19:
Did you make the engraving tests with LW?
After connecting over USB LaserWeb reads the firmware type and version. If this information can not be read from the controller within 10s or is not in the list of supported firmwares, then you will get the "No supported firmware" Error. 

Did you try it with one of the softwares that Sclupfun suggests?

cprezzi, 2022-06-19:
You can use a termianl software (like the serial monitor in the Arduino IDE) to check if the serial connection to the controller theoretically works. Try to send $$ after opening the connection. You should get a list of $ settings back.

ricardofrias, 2022-06-19:
> Did you make the engraving tests with LW? After connecting over USB LaserWeb reads the firmware type and version. If this information can not be read from the controller within 10s or is not in the [...]

I didn't use LW, but I had read in a forum that in case of error when using Lightburn on Mac, I had to use LW first.  This is what I followed and it worked perfectly.

fpronto, 2024-12-05:
Hello
Sorry for open  this discussion. I did also received the same Sculfun S9 model. The laserWeb doesn't seems to be able to connect to it.
LightBurn works fine, but I want to work with open-source software...

I'm using Manjaro Linux. I've opened the a serial connection and the $$ worked fine
Output of $$: 
```
Grbl 1.1h ['$' for help]
$0=10
$1=25
$2=0
$3=4
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=7
$24=200.000
$25=3000.000
$26=250.000
$27=2.000
$30=1000.000
$31=0.000
$32=1
$
$101=80.000
$102=250.000
$10
   $120=1000.000
$121=1000.000
$122=1000.000
$123=200.000
$124=200.000
$125=200.000
$130=410.000
$131=400.000
$132=200.000
```

I'm using sudo to reach the ttyUSB0 connection, and I'm fine with that.
do you have any tips on to make this work?