LaserWeb/LaserWeb4 #644: Wrong serial ports are shown in connection GUI

VanKurt, 2022-07-02:
In the connection tab I can choose between "/dev/ttyS0" ... "/dev/ttyS31". None of these work (laser cutter is not found).

I then checked other tools (e.g. Lightburn, HTerm, PlatformIO), which also can connect to serial ports. These immediately found the laser cutter at "/dev/ttyUSB0".

Is there any way I can use this port in LaserWeb, too? Entering the string "/dev/ttyUSB0" manually seems not possible.

PS: I'm using Manjaro Linux.

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/9033284/Laserweb-Workspace.json.txt)

cprezzi, 2022-07-04:
Your Laserweb-Workspace file doesn't contain the settings that it should. Can you try to export it again?

Are you running the LaserWeb frontend on the same PC?
LaserWeb queries the available ports from the OS (on server side). If you run the server on a different PC, you need to change the `Server IP` under `Comms / Server Connection` to the IP of the server, otherwise you get the ports of the frontend PC.

VanKurt, 2022-07-04:
Hi, I restarted Laserweb and exported the Workspace again.
[Laserweb-Workspace2.json.txt](https://github.com/LaserWeb/LaserWeb4/files/9036862/Laserweb-Workspace2.json.txt)

Regarding frontend/server: I guess so. I downloaded and started "laserweb-builder-v4.0.996-130-x86_64.AppImage". No server involved.

The user rights regarding USB seem fine. As mentioned before, all other applications can access the USB-port just fine (e.g. Lightburn, VS Code etc.). So it seems to be definitely an issue with LaserWeb.

harlock999, 2022-07-04:
Hi,
I'm using Debian and when I type `ls /dev/ttyACM0` I get the following:
`crw-rw---- 1 root dialout 166, 0 Jul  4 09:21 /dev/ttyACM0`

In your case ttyACM0 may be ttyUSB0.

The first important part is `dialout`. This is the group specific for my ttyACM0 port. The second important part is the second `rw` which means the group has read-write access. Thus any user in the `dialout` group would have read-write access to ttyACM0.
When I type `groups` as a user (not root) I see a list of groups I am part of, in my case `dialout` is listed. Now I'm not sure about Manjaro Linux, but in Debian you could type the following:
`sudo usermod –a –G GroupName UserName`
where you replace GroupName by `dialout` and UserName by your user name from which you run the server.

dariusmihai, 2022-10-17:
Hello. I'm also experiencing this issue on Laserweb desktop version.
OS: Windows 11 21H2 build 22000.1098

Laserweb info:
Frontend: 4.0.999
Backend: 4.1.000
Connected to lw.comm-server version 41000 - as reported by version.txt - running on Raspbian 11 (bullseye) on a Raspberry Pi Zero W

My workaround: 
1. Open the desktop app and connect to the server using the Comms page. Only use "Server Connection" at this time.
2 Open the gui in Chrome by navigating to `http://rasp.berry.ip.addr:8000` (Frontend 4.0.999, Backend 4.1.000).
3. Go to Comms page.
4. In the Server Connection section, connect to `rasp.berry.ip.addr` (not localhost, not 127.0.0.1, but the network ip) and click "Connect"
5. In the Machine Connection section, select `USB` and then select the port in the "USB/SERIAL PORT" section and click "Connect"
6. Close the browser and continue using the desktop version as usual, nothing needs to be done on the connection page. Even if the "Machine Connection" section remains empty, Laserweb desktop will work and the console will report that the connection is successful immediately after it is done from the web version.

This way the usb ports are correctly populated, as opposed to the `@undefined` that is displayed in the desktop version.
Since both the desktop and the web versions are connected to the same lw.comm-server instance, it doesn't seem to matter which one triggers the USB connection, they both work as long as at least one of them does the usb connection successfully.

Note: While gathering version info for this message, I noticed that there is a newer version (4.0.999-138) which says that one of the solved problems is "memory leak @ each connect for usb port list". 
I'll post an update once I get the change to test that.

closed 2022-10-17 by cprezzi

dariusmihai, 2022-10-17:
@cprezzi No that's not it. Here is a screenshot:
[image]

I'm already connected to the raspberry pi IP, so I don't think it's a configuration mistake.

cprezzi, 2022-10-17:
@dariusmihai Then you might have used the too new lw.comm-server 4.1.000 which is not compatible with the available desktop app. Only older lw.comm-server is compatible with the app. See https://github.com/LaserWeb/LaserWeb4/issues/649

dariusmihai, 2022-10-17:
Thank you for the clarification @cprezzi, it could be related to that.