LaserWeb/LaserWeb4 #645: gcode out of bounds can lead to running work halt - Elekslaser

gtu78aed87t, 2022-07-29:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

gcode out of bounds can lead to running work halt

I had it working fine until I upgraded to LaserWeb v4.1 on MacOS. 
It connects, but as soon as I go to the "control" tab it disconnects with the message "Warning: gcode out of bounds can lead to running work halt, can lead to running work halt". What can I try to fix or make LaserWeb ignore this?

harlock999, 2022-07-29:
Which v4.1 did you download? The official binary for MacOS looks like v4.0.996
I was about to suggest to load a document first as well. This is not desirable, but a temporary workaround.

gtu78aed87t, 2022-07-29:
I downloaded it from github as v4.1, but it shows up as v4.0.996 when I run it. So it's v4.0.996 I guess.

cprezzi, 2022-07-29:
The latest Mac binary Version named v4.1.000 on LaserWeb4-Binaries combines Backend Version 4.1.000 and Frontend Version 4.0.999 (shown on About page after server connection).

The problem with "out of bound" is mostly caused by a offset of work coordinates to maschine coordinates which is stored into the flash memory of the crontroller. Depending on the setup, you could first run homing and then press the "set zero" button. This should save the home position as origin of the work coordinate system.

closed 2024-05-31 by cprezzi