LaserWeb/LaserWeb4 #648: Missing Installation Instructions

QuirkyCort, 2022-09-27:
In the installation instructions for Raspberry Pi (https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi), under the section "Download and install LaserWeb server", there should be an additional line of...

`sudo apt install libusb-1.0-0-dev libudev-dev`

... before...

`sudo npm install serialport --unsafe-perm --build-from-source`

Without it, the build will fail.

So the complete instructions for that section should be...

```
cd /home/pi
git clone https://github.com/LaserWeb/lw.comm-server.git
cd lw.comm-server
sudo apt install libusb-1.0-0-dev libudev-dev
sudo npm install serialport --unsafe-perm --build-from-source
sudo npm install
```

cprezzi, 2022-09-27:
On which os version did you determine this?
In our tests with buster or bullseye we never needed to install libusb or libudev.  

Did you also read the "more detailed installation instructions" from @easytarget which are linked at the beginning of the page?

QuirkyCort, 2022-09-28:
Sorry. I completely overlooked the @easytarget instructions, so I'm just following the instructions on https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi

I have two installations, one on buster and another on bullseye, both of requires libusb and libudev.

One thing I missed out is that I'm using node 18 rather than node 12. I'm guessing there are no pre-build modules for some dependencies and it needed to be re-build, hence the need for libusb and libudev which isn't required when tested with node 12.

cprezzi, 2022-09-28:
Yes, we are based on node 12 as written in the instructions.
Was it completely working with node 18 after you installed libusb and libudev?

QuirkyCort, 2022-09-28:
Yes. Everything works fine under node 18.

cprezzi, 2022-09-28:
I have added a hint to the documentation about node 18 and the additional libs.

closed 2022-09-28 by cprezzi