LaserWeb/LaserWeb4 #649: Windows frontend cannot connect to Raspberry Pi server

gizmaniac, 2022-09-30:
The latest Windows binary (v4.0.999-138) cannot connect to a Laserweb4 server running on a Raspberry Pi with Bullseye and all latest apt updates.

Opening the server in a browser and connecting does work.  This is not a firewall issue.  Binary searching and test building individual commits reveals that the breakage occurs as a result of commit 30d0379 when the version of socket.io was updated on Dec 26, 2021.

https://github.com/LaserWeb/lw.comm-server/commit/30d0379e872f8651768aa5b5c5519d06d7951698

The prior commit cd58c04 does work.

cprezzi, 2022-10-03:
This is correct. The server already got some API updates that are needed for a new frontend version, but the new frontend is not released yet.

Either use the actual server on the pi and access the frontend over a webbrowser, or use the old lw.comm-server (https://github.com/LaserWeb/lw.comm-server/commit/cd58c04183f3cc59f48a37fb65ee9d1ffd51700a) with the latest exe.

gizmaniac, 2022-10-03:
Thanks for the clarification.  I actually was mistaken in the original post.  The commit that does work is [93aa006](https://github.com/LaserWeb/lw.comm-server/commit/93aa006ba3110838ba8c2fb2616d2133b2dd7c4a), which is the merge to support Bullseye on the RasPi.

cprezzi, 2022-10-03:
Ok, I changed my post accordingly so I don't misslead someone. Thank you.

closed 2024-05-31 by cprezzi