LaserWeb/LaserWeb4 #651: program freezes on gcode generation

jmartens2, 2022-11-06:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

jmartens2, 2022-11-06:
im really sorry. my issue doesnt allow me to save after gcode generation. basically the program freezes at 100% of generation but doesnt offer the tabs needed to move file or any thing. ive tried different file formats, changing settings, shutting down my macbook pro 2013 with 8 meg ram, once or twice if i reset to factory itll work again. today isnt one of those days. could you please give a guy a hand

jmartens2, 2022-11-06:
i sorry about the open and close of issue. i must read instructions better.

harlock999, 2022-11-06:
I'll do my best to help. What do you have in your GCODE setup? Which GCODE function are you trying to perform?

jmartens2, 2022-11-06:
marlin

harlock999, 2022-11-06:
I have no experience with marlin but I am not sure how this can affect the GCode compilation.
- Can you try with a different and very simple file?
- Did you recently performed an upgrade on nodejs?
- Did you try to hard-clear your browser cache?
- Do you have any hardware diagnostics you can run on your computer ? (sorry if this sounds lame but it's a laptop almost 10 years old...)

harlock999, 2022-11-06:
Also, when I was asking about GCODE setup, I meant the TOOL ON and the like. There's a known issue when combining INTENSITY and some Laser raster functions. For now I believe INTENSITY should only be used with Laser Cut.

jmartens2, 2022-11-06:
yes ill try different file
im unfamilar with nodejs
i will try clearing cache
ran first aid this morning
tool on is m106 p0
laser intensity is set to S

jmartens2, 2022-11-06:
simple jpg generates
i found that in tool head beam was set to 0.2mm the program gives message that 0 and 1 are closest values. could this be problem?

harlock999, 2022-11-06:
0.2mm is quite reasonable value, but I vaguely remember something about some similar message in the past. what happens if you have it as 1 (just for testing)?

harlock999, 2022-11-06:
Also if you are using a simple JPG, I'm assuming you are doing raster engraving of some sorts?

jmartens2, 2022-11-06:
i think beam size in machine settings has to match in gcode laser diameter? and it has to be 1 or 0. i will update.
thank you so much for your help so far i love this program

jmartens2, 2022-11-06:
thank you again. as long as i leave it at 1 it will generate gcode.
as i burn the project it leaves the lines too far apart, is the a way to bring closer together? horizontal lines.

harlock999, 2022-11-06:
Something is going on with the error checking as 0.2mm would make the lines closer. Can you try to put 0,2 instead of 0.2mm? I have seen at times on other software that the decimal dot or comma aren't interpreted correctly.

jmartens2, 2022-11-06:
i shall try. what should the beam size be in machine settings?

harlock999, 2022-11-06:
It depends on your machine, optics (distance between focal point and stock material) etc. I believe 0.2mm is a good starting point.

jmartens2, 2022-11-06:
0.2 crashes at 97%
no but i can google

harlock999, 2022-11-06:
If you have something weird in your GCode setup the GCode processor may not like it. You may have an extra space or character at the wrong place. You can try to create a new machine profile and carefully re-enter your settings.

jmartens2, 2022-11-06:
as long as i keep the laser diameter at 1 it generates.
i will create a new profile. 
i hate to keep you from doing things but thanks

jmartens2, 2022-11-06:
im trying to make a new machine profile and the old settings are populating the boxes

harlock999, 2022-11-06:
oh? I don't recall this behavior, but then it's been a while I haven't tried myself. Make sure all the GCode settings are cleared and retype them.

jmartens2, 2022-11-06:
working on it.....

jmartens2, 2022-11-06:
this is strange. as long as i leave laser diameter on 1 it generates a gcode. im going to set this down for a while and pick up later. time to do my chores according to the boss of this house. 
thank you so much for your help so far!!! back in a while take your time to enjoy your day!

harlock999, 2022-11-06:
np, I believe next step is opening the developper console in the browser and see if you get any interesting message there.

cprezzi, 2022-11-07:
The "TOOL HEAD / Beam diameter" just tells LW how big a pixel should be and how far the scanlines should be from each other. This should match about the width of an engraved line. 0.2 is a good value for a CO2 laser.
(Don't use the up/down buttons of the Beam diameter filed, as they only increase/decrease by 1)

This value also defines how much data will be created. For example, if you have a 100mm by 100mm picture and 0.2mm tool diameter, it will create a point for every 0.2mm which results in 500 lines x 500 points (= 250'000 points).
An image of 200 x 200mm would already produce 1000 x 1000 points (= 1'000'000 points)! 
This is the reason why bigger pictures can reach the available memory limit.

jmartens2, 2022-11-07:
Those are some of the things that I’ve been trying. I’m currently down to .3 mm on laser diameter. It takes a little bit to generate. Is there a way to elevate the available memory that the program can access or is that tied to my machine (laptop)?

jmartens2, 2022-11-07:
Oh my manner are terrible. Thank you for your help here on this project. You’ve given me a ray of hope, I’ve been floundering in info on the internet about engraving and no real help. You guys are amazing.

jmartens2, 2022-11-07:
Hi I’ve been able to get to .2 mm on my laser diameter. It takes a few minutes to get there but I can live with that. 
another question, does using the filters for the generation affect the gcode generation?
Say greyscale, does it add to problem?
Alison during these trials I’ve realized my beam setting in machine is set to .1 will this also make the program work harder than if it’s different from laser diameter or do they provide separate results on generation?

cprezzi, 2022-11-07:
I don't now if the filters make the memory problem worse or not, but I usually prepare the image in a photo editig software anyway to convert to grayscale, optimize contrast and adjust the resolution to match the needed pixels. This way you don't provide a larger image to LW which uses more memory and could create aliasing in the scale down process.

The "Beam diameter" from the machine settings is the default value that is copied to new "Laser Raster" operations. Only the value in the operation is used for the gcode generation.

jmartens2, 2022-11-07:
Thank you for all the hard work on this project. 
by needed pixels, how does one come to that number to be used?

cprezzi, 2022-11-08:
You can calculate like this:
Size you want in mm divided by laser diameter setting = pixels

Example: if you want the engraving to be 200 by 100 mm and use 0.2mm laser diameter, you would need a picture with 200/0.2=1000 by 100/0.2=500 pixel.

cprezzi, 2022-11-08:
There is some documentation on our website: https://laserweb.yurl.ch/documentation/cam-operations/40-laser-operations/8-laser-raster-settings
And tere are lots of infos on google and youtube from other people about picture engraving (not specific to LaserWeb).

jmartens2, 2022-11-08:
Thank you again

jmartens2, 2022-11-10:
Thank you all for your help. I’ll close this one. Enjoy life, be safe, and keep burning!!!

closed 2022-11-10 by jmartens2