LaserWeb/LaserWeb4 #653: Features request

remi75, 2023-04-01:
Hi
Possible to have an option to specify number of passes aka redo the CNC job N times ?

Regards

harlock999, 2023-04-01:
Hi,
Isn't it there already? For laser operation, there's a Passes operation just before cut rate.

Can you elaborate what you are trying to achieve?

Cheers!

remi75, 2023-04-01:
I do not see it ...
laserweb 4.0.996 under Linux
top left of control i see: X, Y, F and S only
firmware grbl 1.1f on my laser engraver
yes passes as run the whole cnc job file X times.
many users with weak lasers need it, or even for specific situations...
:)

remi75, 2023-04-01:
I see, I dont have that option because I load a g-code file, not a graphics file (dxf or svg etc)
In that case , is this feature interesting ?

harlock999, 2023-04-01:
Loading the G-code is a different thing. You see, the passes setting is for generating a new G-Code. Once the G-Code is generated, you can manipulate with a text editor such as Mousepad in Linux, you can copy/paste the section you need to be repeated, but you need to understand what is going on with the G-Code. Very often there will be a header to set the machine properly and at times some more lines at the end to turn off auxiliary devices and put the machine in a relative safe state.

But here what you are asking for Laserweb to interpret back the G-Code and figure out which section needs to be repeated. I believe that so far the G-code generator is a one-way process. Once it is prepared, it is sent to whatever controller firmware you have, in this case grbl. But it doesn't interpret it back into a shape, only thing Laserweb knows is position and some basic states of the controller.

Are you generating the G-Code from outside Laserweb? I'm asking because I feel it's more straightforward to save the Workspace, reload it when needed, adjust the number of passes and generate it for each time a new engraving or cutting is needed.

remi75, 2023-04-01:
:)
ok, I guess I need like a maro: that loads the gcode execute it, put the laser back at 0.0 and do it again ...
oh, going to learn another computer language again (gcode and its macros) ... sniff.
If some reader is already familiar with it... I buy it !

cheers

harlock999, 2023-04-01:
Well if that's all you need you may want to look at putting a simple command G0 X0Y0 at the end of your GCode. Be aware of other commands that may turn off ventilation, laser cooling or even motor drivers enable/disable. They may or may not need attention.

Have you tried not doing anything special and just rerunning the G-Code? Unless there's something special with your setup, once your origin point is set, it shouldn't matter where your laser is at the beginning of the job, it will always move to the same (x,y,z) coordinates.

Bonne chance!

remi75, 2023-04-01:
yes it works, but automation needed if I need 20 passes, it is not practical to manually restart 20 times ...

harlock999, 2023-04-01:
aaah that's quite an appreciable number of passes

harlock999, 2023-04-01:
Tentatively with a brute-force approach for a one-of file, I'd copy-paste the G-Code chunk of interest with Mousepad to get your 20 number of passes. Try to see what is the header and footer of your file and keep those intacts without copying them.

If you don't mind, it's probably best to move this conversation over to the maker forum.

https://forum.makerforums.info/c/cad-cam/laserweb-cncweb/78

remi75, 2023-04-01:
ok, thank you.

closed 2023-04-01 by remi75