LaserWeb/LaserWeb4 #655: Clipped (cropped) Inkscape SVG files do not maintain the clipping in LW4

awmt102, 2023-04-27:
When i apply a clip (crop) to an object in Inkscape it clips correctly. When I import this into LW4 the clip is not pulled through and the entire unclipped element is visible. Example files attached.

LaserWeb 4.0.999 on Mac OS Ventura 13.2.1
Inkscape 1.2.1

[test.json.txt](https://github.com/LaserWeb/LaserWeb4/files/11342974/test.json.txt)
[image]
[image]

cprezzi, 2024-05-31:
If something is shown in LaserWeb, it obviously was in the svg. So your croping didn't work.
Instead of croping, you could try to copy only the needed parts to a new file.

easytarget, 2024-07-12:
https://forum.makerforums.info/t/clipped-cropped-inkscape-svg-files-do-not-maintain-the-clipping-in-lw4/88040
and
https://forum.makerforums.info/t/clipped-cropped-inkscape-svg-files-do-not-maintain-the-clipping-in-lw4/88040/4

Are essentially the same topic; LW takes a very simple approach and generates paths for all lines and curves in the source. It has no concept of 'clipping', the above topics cover this and how to work around.

closed 2024-07-12 by easytarget