LaserWeb/LaserWeb4 #659: Jerky motion due to too many instructions

T-Workshop, 2023-07-16:
Hi,
I am dealing with an issue that I come across at different forums. When raster engraving (or raster merge) the laser becomes very jerky at higher speed (200 mm/s).
After a bit of a headache I tried generating gcode for the same image with LaserGRBL and that gcode ran fine with the same speeds and accels. After I compared the gcodes I came to the conlusion that LaserWeb is "too good" and for the M4 dynamic power it generates too many steps of the power. To simplify-where LaserGRBL steps down the S value by "1", LaserWeb steps by "0.1". 

Is there any way to reduce the "resolution" of the dynamic power mode?

Thanks
Tomas

easytarget, 2023-07-17:
Please look at the 'Segment' option in the job options. This controls the resolution of the gcode produced.
It defaults to `0`, aka maximum resolution. But setting it to 0.2 or so will speed things up without affecting output quality.

We recentl discussed this in the official support forums on MakerForums.info 
[Please see this post, it has examples etc.](https://forum.makerforums.info/t/too-fine-resolution-when-creating-gcode/87431/15?u=easytarget)

T-Workshop, 2023-07-17:
Thanks for your reply.

Sadly this problem occurs while rastering and (please correct me if I'm wrong) there is no "segment" option there.

easytarget, 2023-07-17:
Sorry of the initial mis-read, I've fixed the original answer

closed 2024-05-31 by cprezzi