LaserWeb/LaserWeb4 #661: Changes to frontend not reflected

polchky, 2023-08-18:
Hi,

I am trying to modify the project in order to include new macros for replacing the plunge and retract operations of the mill operations by custom Gcode commands (to ultimately control a servo instead of a spindle). I thought the easiest way to test these changes would be to build and launch a dev docker image as described in the README: 
`docker build --target dev -t laserweb:dev .`
and then 
`docker run -it -device=/dev/ttyUSB0 --rm -p 8000:8000 laserweb:dev`

However, I noticed that the frontend (http://localhost:8000) did not contain the changes I made. I then tried to change purely graphical elements (such as titles), destroyed the container and the image and built everything again, but still without seeing any change. Did I miss something? Do I need to build the frontend in a different way? I though that the Docker container would do the needed job. 

Thanks a lot in advance

harlock999, 2023-08-18:
Hi,
Have you tried a hard clearing of the cache? In Chrome, open the developper tools Ctrl-Shift-i then Ctrl-rightclick on the Refresh icon or Ctrl-Shift-R for a hard clearing of the cache.

Regarding plunging, I'm using more or less a laser cutting but with plunging capability, I've added simply custom GCode at beginning of the path and using simple laser cutting functions. Works like a charm. However I don't need the other cutting or engraving functions, so probably you'd need a different machine profile for each of the tasks if you're switching between different GCode generators.

polchky, 2023-08-18:
Thanks for the suggestion,

I have tried hard clearing the cache as you suggested, without luck. 

Unfortunately for my use case I cannot simply use the same solution you propose, since I need the macro to be applied after the tool head moved to position for the start of each operation (I'm controlling a pen and using a custom plotting device).

harlock999, 2023-08-18:
I'm not sure I understand. I used a pen as well to check operation before mounting my laser head as a proof of operation. There are macros that can be applied before *each path* and also before *each job*. If your path is made of a polyline of some sort with vertexes it should work.

polchky, 2023-08-18:
The problem is that the existing features are not enough for what I want to do. I want to both outline and fill a shape using a pen (and thus an up-and-down motion before and after each path). The "Laser Cut" operation works for the outline using the "Tool On" custom Gcode in the options. However, using the "Laser Fill Path" operation to fill the shape results in the up-and-down motion for each single straight line inside the shape, which results in a significantly longer work time. And I cannot use the "Mill Pocket" operation either, since the "Tool On" option is not applied in this case. There is a "Pre Op" macro, but it is applied before the tool head moves into position, which would result in an extra drawn line between the current position and the start of the next path.

Concerning the initial problem I tried to build the Docker image and to deploy it on another machine to be sure that the caches were not the issue, but still without success...

harlock999, 2023-08-18:
I just realized that you specifically mentioned docker. I don't use docker, I run straight from Debian bookworm these days. Because  the fanless computer I'm using is fast enough, I rebuild the front-end every time.
Have you tried to use `laserweb4:dev-es6` instead of `laserweb:dev` ?
There is a specific command to rebuild with npm. I use `npm run start-dev` for the front-end, and use `npm start` for the server only.

cprezzi, 2023-08-21:
I would suggest a completely different approach.
In the backend server (server.js of lw.comm-server) from line 2307 you find the function runJob which parses each gcode line befor adding it to the queue. There you could search for the S param (Laser Power) and add your up/down gcode for the servo before or after the parsed line.

polchky, 2023-08-21:
Thanks for the advice!
I think my main confusion was the separation between frontend and backend, and how to build each part. Since I want to do something parametric (with instructions for the servo that could be customized) I would need to change at least the frontend. I ended up choosing another solution altogether and I wrote a simple JS web client to convert a gcode file, more specifically the plunge and retract codes from the generated paths for the "Mill Cut" and "Mill Pocket" operations (which resemble my use case the most), and also removing the extra "retract" code at the beginning of each operation.
This way I can also keep using the latest version of Laserweb :)

In any case thanks for the responses and for the great job on this project!

closed 2024-05-31 by cprezzi