LaserWeb/LaserWeb4 #662: Y axi is opposite

Sagigamil, 2023-09-15:
Hi
I try to use laserWeb with 3d printer with laser head addon. I noticed that y axis is oposite:
When homeing the laser headset comes to the right position (0,0) but showed on softwar as (0,230). Also the direction button of y movement is different. When I move this axi using gcode/cura it moves to right direction.
What should I do?

Thanks

cprezzi, 2023-09-21:
Where is your homing location relative to the bed? (top/left, top/right, bottom/left, bottom/right)?
Does the head or the bed make the y movement?

harlock999, 2023-09-21:
Which firmware is being used? How does it work with the 3D printing software?
Assuming your 3D printer works for 3D printing tasks, I'm not clear why Y axis would be backward with Laserweb.
In Smoothieware, you can set homing position, see
[(https://smoothieware.org/configuration-options)] with variables `beta_homing_direction`, `beta_min` and `beta_max`

But what is strange is homing works, so it makes me believe the Y axis connections between the controller and stepper driver are ok? Unless both homing position and Y direction signals are swapped.

Also, careful with position being shown in Laserweb. These are workpiece coordinates, not machine coordinates.

Sagigamil, 2023-09-21:
> Where is your homing location relative to the bed? (top/left, top/right, bottom/left, bottom/right)? Does the head or the bed make the y movement?

buttom left (if I look from top view) but in the app is shows as top left

Sagigamil, 2023-09-21:
> Which firmware is being used? How does it work with the 3D printing software? Assuming your 3D printer works for 3D printing tasks, I'm not clear why Y axis would be backward with Laserweb. In [...]

I use marlin, and when I use the screen to move the axis everything is fine, I'm familiar with 3d printing and I'm sure that it's ok in this matter. 
I see that the beam looks like it's from button and not top in the app so I think that something is wrong there, like it's mirrored somehow

cprezzi, 2023-09-21:
LaserWeb shows the head at 0,230, because that's what your printer is reporting in the position messages after homing.
So it seems that a work offset of 0,-230 is stored in your firmware. 

Just click on the "set zero" button. That tells the firmware, that the actual position should be 0,0.

closed 2024-05-31 by cprezzi