LaserWeb/LaserWeb4 #663: Where do I get the frontend for Linux

SidSkiba, 2023-11-09:
I built the latest server on Linux Mint 21.2. I followed the info on [https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi](https://laserweb.yurl.ch/documentation/installation/36-install-raspberry-pi) and it works properly. There are some warnings and such during the build but it builds fine with latest NPM and NODE. I can access the server using Chrome and localhost:8000. It connects to the Laser and I can control it fine. 

I however want to use the front end on the same machine without using Chrome. 

I then tried the Appimage provided on [https://github.com/LaserWeb/LaserWeb4-Binaries/](https://github.com/LaserWeb/LaserWeb4-Binaries/). This also worked good and included the front end. But I was unable to connect to the laser. It would connect then state no supported firmware found and disconnect. I assume the Appimage cannot connect properly to my Ortur LM2 S2 machine (GRBL). As the lw.comm-server works fine I am sure I have the dialout setup correctly. 

Wondering if anyone has way to compile LaserWeb4 the front end as described here: 

Install the frontend
Install the LaserWeb4 executable on your Windows, OSX or Linux PC. This way you get the lastest frontend version.

    Connect your machine to one of the USB ports
    Start the LaserWeb4 app
    Click on Comms tab
    Click on "Server Connection"
    Change Server-IP to `YOUR-SERVER-IP:8000`  (replace YOUR-SERVER-IP with the IP of your LaserWeb Server)
    Click connect
-> You should see a green message "Server connected" in the log area at bottom right.

and hinted at here:

https://gist.github.com/easytarget/fc7a6dabf893d1b461c5e66573d0725a

Regards,

Sid

harlock999, 2023-11-09:
Hello,
LaserWeb4 was specifically designed to work with Chrome, See note below the first screenshot in the link below:
[(https://github.com/LaserWeb/LaserWeb4/wiki)]

So with that in mind, you may find unsupported features or unexpected behavior with other browsers. This may have serious effects (damage? injuries?) when controlling a laser or any type of machinery.

That being said and assuming you can take proper safety measures, have you tried opening another browser when the backend was running in the first install you described with Linux Mint?

SidSkiba, 2023-11-09:
Thank you. I may be misunderstanding how to do this on Linux. I was using LaswerWeb on Windows 10 and loved it. I now want to move my PC for the Laser over to Linux because Win 11 will not be supported on it. 

In any case I installed the backend on the Linux Mint and could access LaserWeb with Chrome web browser in Linux Mint. It seemed to work correctly (though I have not tried a real burn with it). 

I am far from a Linux wizard. So I wanted to use LaswerWeb the way I did on the PC. On Linux it seems I have to enter the terminal, launch the server, then open Chrome and access Localhost:8000. 

I wanted to (in Linux) simply launch LaserWeb from the desktop like Windows. I assumed with the front end application would do that. I launch the front end and it launches the backend and then closes the backend when it closes.

Is there a way to do that in Linux with the browser? Can I launch the backend, launch Chrome, then close backend when I shut down. As this is not dedicated to the laser I did not want to have the backend running all the time. Only when I am using it.

harlock999, 2023-11-09:
The front end and back end are two different processes. Depending how you launch LW4, the browser is also launched automatically.

Based on the raspberrry pi wiki, it seems the backend is used as a service through systemd using `systemctl start lw.comm-server` or `systemctl enable lw.comm-server` to have the backend start automatically upon startup. If you want to shutdown the server, you can run `systemctl stop lw.comm-server`

Now the usage of a service with systemd is entirely optional.
Note that you can launch multiple programs simultaneously, you can create a script in linux, the equivalent of a batch file in DOS / Windows.
To run commands at the same time, add a `&` at the end of the command.

Have a look at the following link: [https://github.com/LaserWeb/LaserWeb4/wiki/How-to-compile]
I haven't tried those steps with the lastest versions of npm and node.js. I always recompile every time I start LW4, I know next time I make a small modification or a dependency upgrade I will forget how to do it.

SidSkiba, 2023-11-10:
I think I have it figured. I do not need to compile the front end as it is already in the server. All I need to do is build the backend then drop the following script into lw.comm-server :

```
#!/bin/bash

# Function to check if the Chrome browser is running
is_chrome_running() {
    pgrep -x "chrome" > /dev/null
}

# Start the Node.js server
node server.js &

# Store the PID of the last background command (Node.js server)
server_pid=$!

# Launch Chrome browser pointing to localhost:8000
/usr/bin/google-chrome-stable --app=http://localhost:8000

# Check if Chrome is still running
while is_chrome_running; do
    sleep 1
done

# Terminate the Node.js server
kill $server_pid
```

It seems to work correctly. I have not tried it yet but will. If it works I will document the steps to get it running on Linux Mint. Should be pretty much the same for all debian cases.

SidSkiba, 2023-11-11:
**OK I have it working in Linux Mint V21.2 Cinnamon Edition.**

The installation prcess was as follows.

Open the terminal window and enter:

```
sudo apt-get update
sudo apt install npm
```

You can verify that npm and node are installed then by entering the following:

```
npm -v 
node -v 
```
The following were the version number returned at the time of this writing:

npm - 8.5.1
node - 12.22.9

Then install Chromium by entering:

`apt install chromium`

Install git:

`sudo apt-get install git`

Clone and build the latest lw.comm-server:

```
cd /usr/local
sudo git clone https://github.com/LaserWeb/lw.comm-server.git
cd lw.comm-server
sudo npm audit fix
sudo npm install 
```
You can then test that the sever is working with:

`node server`

Type CTRL-C to stop the server: 

Then create a bash script for launchng the server and the browser:

`sudo nano startlw.sh`

Copy and paste or type the following into the nano editor.  Note that this script can be used with Chromium (recommended) or Firefox by uncommenting the correct line:

```
#!/bin/bash

# Start the Node.js server in the background
node /usr/local/lw.comm-server/server &

# Store the PID of the last background command (Node.js server)
server_pid=$!

#ONLY UNCOMMENT ONE OF THE FOLLOWING BROWSERS
# Uncomment the next line to launch Chromium browser pointing to localhost:8000
chromium --app=http://localhost:8000/ --start-maximized
# Uncomment the next line to launch Firefox browser pointing to localhost:8000
#firefox -new-window http://localhost:8000/

# Terminate the Node.js server
kill $server_pid
```

Type CTRL-O then CTRL-X to save the script and exit nano editor.

Then make the script executable and change the owner to your username (replace all the "doug" below with your own Linux username):

```
sudo chmod u+x startlw.sh
sudo chown doug:doug startlw.sh
sudo usermod -a -G dialout doug
```

You should be able to launch LaserWeb fro the terminal then by entering:

`./startlw.sh`

You are ok to close the terminal if everything worked. 

You then can add an icon to your desktop by right clicking on the desktop and picking "Create a new launcher here" from the context menu.

For name use: LaserWeb4
For command use: /usr/local/lw.comm-server/startlw.sh

If you want to change the desktop icon there is one included in usr/local/lw.comm-server/build/ folder.

cprezzi, 2023-11-13:
@SidSkiba Thank you very much for the nice instructions. Is it ok for you when I add these instructions to the wiki?

SidSkiba, 2023-11-13:
I made a few edits above for spelling and clarity on dependencies. I cannot seem to do a pull on master?