LaserWeb/LaserWeb4 #664: No supported Firmware Detected on Linux AppImage

dvilarrocha, 2023-11-28:
Hi,
Firts of all thaks for developing Laseweb4 which is a great and useful application.

I have installed Laserweb4 on Windows and Mac and I am successfully using it with my Laser Cut "Leon 3D L40", which works with grbl v1.1f firmware. https://www.leon-3d.es/product/l40/

On Linux Ubuntu, I have tried with the App Image and I always get the same message "No supported Firmware Detected ". I have tried to add my user to dialout and tty grpups following the steps on this website https://hackaday.io/project/19103-the-technique-of-laser/log/185061-connecting-usb-lasers-to-linux. 
I have also read other posts in this forum related to the .env file, but with the AppImage version I think it doesn't work.

Any Idea about the problem? Linux is necessary for us because in our high school all the computers run with Linux and we prefer not to use our personal laptops.

Thank you.

cprezzi, 2023-11-28:
As you probably realized, the App Image Version is old. Insted you can use the last lw.comm-server which has a integrated webserver with the frontend. So you can use LaserWeb with the Chrome/Chromium Browser. See https://github.com/LaserWeb/lw.comm-server/wiki/Manual-Installation-(Linux) for installation instructions.

dvilarrocha, 2023-11-30:
Thanks cprezzi.

I have installed it and works fine starting manually the server with "node server.js", but when I Setup Systemd to automatically start the server I get the the following error message for "sudo systemctl status lw.comm-server":

sudo systemctl status lw.comm-server
● lw.comm-server.service - LaserWeb4 server
   Loaded: loaded (/home/administrador/lw.comm-server/lw.comm-server.service; enabled; vendor preset: enabled)
   Active: **failed** (Result: exit-code) since Thu 2023-11-30 08:46:45 CET; 7s ago
  Process: 3541 ExecStart=/usr/bin/node /home/pi/lw.comm-server/server.js (code=exited, status=217/USER)
 Main PID: 3541 (code=exited, status=217/USER)

de nov. 30 08:46:45 LliureX-19-Client systemd[1]: lw.comm-server.service: Service hold-off time over, scheduling restart.
de nov. 30 08:46:45 LliureX-19-Client systemd[1]: lw.comm-server.service: Scheduled restart job, restart counter is at 5.
de nov. 30 08:46:45 LliureX-19-Client systemd[1]: Stopped LaserWeb4 server.
de nov. 30 08:46:45 LliureX-19-Client systemd[1]: lw.comm-server.service: Start request repeated too quickly.
de nov. 30 08:46:45 LliureX-19-Client systemd[1]: lw.comm-server.service: Failed with result 'exit-code'.
de nov. 30 08:46:45 LliureX-19-Client systemd[1]: **Failed to start LaserWeb4 server.**

In other PC I get the message "too many levels of symbolic links" when enabling and starting the service.

Any idea for these errors?

Thank you.

dvilarrocha, 2023-12-01:
I have solved it. It was a problem of updates and users.

closed 2023-12-05 by cprezzi