LaserWeb/LaserWeb4 #668: laser test button

faisalzahid12, 2024-04-10:
the laser test button is not working when I click the button it does not turn on and if I click on check size laser steel won't turn on but when I start the job then the laser turns on and cuts the material I am having a hard time and wasting material because I can't tall ware the laser is please hele'

cprezzi, 2024-04-10:
By default, the "TOOL TEST POWER" (in settings/gcode) is set to zero for safety reason.

- If you have a diode laser (visible beam) then you can set "TOOL TEST POWER" to a low value so it does not burn but is visible. Additionally set "TOOL TEST DURATION" to 0 for unlimited duration.
- If you have a CO2 laser (invisible beam) you should only use this function for a short pulse. Set a low "TOOL TEST POWER" and a short duration like 100ms in "TOOL TEST DURATION", so that it only marks a point on your material.

cprezzi, 2024-04-10:
The "check size" function does not activate the laser (for safety reason.

- On a diode laser (after configuring tool test power and duration), you can first click "Laser Test" and then "Check Size" to see the outline of your job. Click "Laser Test" after this again to switch the laser off.

faisalzahid12, 2024-04-10:
then you for the advice but can you give me some stap-by-stap guides I will be grateful

cprezzi, 2024-04-10:
I have no idea what machine you have, but many chinese vendors tell you it's 40W, but in reality it is only 5W continous. The cutting power is primarily depending on laser power, speed and focus. Lower speed results in more cutting power, bad focus means less power. If your material is not flat, then focus is not constant and so is power.

faisalzahid12, 2024-04-11:
how to check the true watt of my diode laser

faisalzahid12, 2024-04-11:
I tested a cut speed on a 3mm strip my x =280 and 150 and1 pass power 100% for some reason (x) has thean line but (y) has the line
[image]
![Uploading 20240411_222715[1].jpg…]()

faisalzahid12, 2024-04-11:
![Uploading 20240411_222715[1].jpg…]()

faisalzahid12, 2024-04-11:
[image]

faisalzahid12, 2024-04-11:
[image]

faisalzahid12, 2024-04-11:
[image]

cprezzi, 2024-04-12:
There are Laser Power Meters, but they are very expensive. For example: https://www.cloudraylaser.com/products/cloudray-0-60w-high-accuracy-laser-beam-power-and-energy-meter?_pos=3&_sid=eb92f5c36&_ss=r

The beam of a diode laser is not round, it is rectangular. It has different dimentions in x and y direction. The result on your picture is normal.

But this is the wrong place for such questions. This is the issue tracker for LaserWeb!
Please serach for a general lasercutter forum. I don't have time to be your general supporter ;)

faisalzahid12, 2024-04-12:
ok thank you for all your help good bay

closed 2024-04-12 by faisalzahid12