LaserWeb/LaserWeb4 #670: laser starting point

faisalzahid12, 2024-04-21:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

faisalzahid12, 2024-04-21:
is there a way to tall the laser which points to the cart cutting

harlock999, 2024-04-21:
In CNC world, there's mainly two coordinates systems: machine coordinates and workpiece coordinates. LaserWeb shows the workpiece coordinates, so when you click Set Zero in the Control tab it will zero out all three coordinates. After that, it depends how your original file and generated GCode is built.

closed 2024-04-22 by faisalzahid12