LaserWeb/LaserWeb4 #672: how to tell laser to cut inside or outside the line in laser web

faisalzahid12, 2024-05-31:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

faisalzahid12, 2024-05-31:
how to tell the laser to cut inside or outside the line in laser web

cprezzi, 2024-05-31:
Just select the corresponding operation "Laser Cut Outside" or "Laser Cut Inside" and set the Laser Diameter to the kerf  width (width of cut that your laser is making).

closed 2024-05-31 by cprezzi

faisalzahid12, 2024-06-05:
can you give step by step plese