LaserWeb/LaserWeb4 #674: Retry the laser status remotely

magostinelli, 2024-11-14:
Anyone knows if is ti possible to know the engraving process status remotely? Like an API top call, or otherwise a command from ssg that return the current status?

Thanks.

cprezzi, 2024-11-14:
There is only a qCount Event sent from server to client, but this does only tell how many gcode lines are in the server queue. This is not exact, because the firmware also has an own queue. See https://github.com/LaserWeb/lw.comm-server/wiki/API-(for-client-communication)

cprezzi, 2024-11-14:
PS: You can connect to the server from multiple clients at the same time.

magostinelli, 2024-11-14:
Thanks for reply, in the link posted there is runStatus how can i use it?

I'm looking at something like thie:
curl -s http://laserweb.local/status -> idle|ingraving|stopped

cprezzi, 2024-11-14:
There is no http API. You need to connect to the server by websocket (socket.io). Unfortunately I can't help with that. You eventually can find out by reviewing the lw frontend code.